Enapter/marketplace #4: Add section "Connect to Enapter" to blueprint's readme files

kulti, 2021-12-09:
At now blueprint's readme files contain phrase like "Use Enapter ENP-RS485 module for physical connection". Think it should be another section like this:

## Pylontech US200 (RS485)
This _Enapter Device Blueprint_ integrates **Pylontech US2000** - lithium battery control and monitoring via [RS-485 communication interface](https://developers.enapter.com/docs/reference/ucm/rs485). It also supports CAN bus.

### Connect to Enapter
Use [Enapter ENP-RS485](https://handbook.enapter.com/modules/ENP-RS485/ENP-RS485.html) module for physical connection.

### RS-485 communication interface parameters
+ Baud rate = `9600` bps
+ Data bits = `8`
+ Parity - `N` (no parity)
+ Stop bits = `1`

### References
+ [Pylontech US2000 product page](http://en.pylontech.com.cn/pro_detail.aspx?id=114&cid=23)

nikitug, 2021-12-13:
Also just discussed with @anataty that we would remove communication interface parameters from the readme and document it in the source code instead.

nikitug, 2021-12-14:
The final template:

- Sign up using [Web](https://cloud.enapter.com/) or mobile app ([iOS](https://apps.apple.com/app/id1388329910), [Android](https://play.google.com/store/apps/details?id=com.enapter&hl=en)).
 - Use [Enapter ENP-RS485](https://handbook.enapter.com/modules/ENP-RS485/ENP-RS485.html) module for physical connection. See [connection instructions](https://handbook.enapter.com/modules/ENP-RS485/ENP-RS485.html#connection-examples) in the module manual.
 - [Add ENP-RS485 to your site](https://handbook.enapter.com/software/mobile/android_mobile_app.html#adding-sites-and-devices) using the mobile app.
 - [Upload](https://developers.enapter.com/docs/tutorial/uploading-blueprint/) this blueprint to ENP-RS485.

closed 2021-12-14 by nikitug