Enapter/marketplace #271: [Mass Flow meter, Alicat, and Whisper mass flow meter]

tjzay, 2023-11-02:
**Device Description**
Provide a clear and concise description of the device, and include the link to the device model on the vendor website.
Low pressure mass flow meter. https://www.alicat.com/models/mw-low-pressure-drop-gas-mass-flow-meters/

**Device Access**
Do you have access to this device? Will you be able to test the integration?
Yes, device has been ordered and will be available for testing.

**Communication Protocols**
Specify the communication protocols that the device supports. E.g. Ethernet, RS485, RS232, CAN, DIO, etc.
Modbus RTU and ASCII over RS-485 communication

**Documentation**
Include publicly available links to the device documentation. If there are no public links available, please attach the documentation file to the issue.
[DOC-MANUAL-MPL.pdf](https://github.com/Enapter/marketplace/files/13233586/DOC-MANUAL-MPL.pdf)
[DOC-SPECS-MW-MID-2.pdf](https://github.com/Enapter/marketplace/files/13233587/DOC-SPECS-MW-MID-2.pdf)

**Questions and Ideas**
If you have any integration questions or implementation ideas that will ensure a high-quality integration, please suggest them here.

anataty, 2023-11-03:
Hi @tjzay! Thank you for the links. I checked documentation and looks like Whisper Series can be covered by our [Alicat mass flow meter Models M · MC · P · PC · L · LC blueprint](https://github.com/Enapter/marketplace/tree/main/flow_meters/alicat_flow_meters_rs485_modbus). Could you check it out and tell me if it works for you?

tjzay, 2023-11-14:
Hi @anataty. I did see this before I uploaded but thought that it wouldn't necessarily work as the blueprint does not list MW (whisper series) as a model. If you think it will work anyway, I will test it once it arrives using this blueprint. Thanks!

anataty, 2023-11-17:
It should, because Alicat shares the same manual ([here](https://www.alicat.com/models/mw-low-pressure-drop-gas-mass-flow-meters/) under **Learn More section**) as for M / MC etc. models. In case of any problems - you can write here or in [our Discord](https://discord.com/invite/TCaEZs3qpe) :)

tjzay, 2023-11-20:
Ok thanks a lot!

tjzay, 2024-01-30:
Hi @anataty,
I just realised I require a virtual UCM with serial communication using Arrakis Mk4. Does the existing blueprint which uses RS485 UCM work with virtual UCM too?

nkrasko, 2024-02-05:
@tjzay it is required to adopt this blueprting to work with Arrakis MK4 same as it was done for [Crowcon Xgard Bright](https://marketplace.enapter.com/device/crowcon-xgard-bright/crowcon_xgard_bright_arrakis_mk4). We will work on this.

closed 2024-02-06 by nkrasko