Enapter/marketplace #274: [Rain sensor, Seven Sensor, and 3S-RG]

tjzay, 2023-11-20:
**Device Description**
Provide a clear and concise description of the device, and include the link to the device model on the vendor website.
This is a rain sensor, https://www.sevensensor.com/rain-gauge-with-modbus-rtu-output.

**Device Access**
Do you have access to this device? Will you be able to test the integration?
Yes, it is currently being shipped.

**Communication Protocols**
Specify the communication protocols that the device supports. E.g. Ethernet, RS485, RS232, CAN, DIO, etc.
RS485
Baud: 9600. Parity: even. 1 Stopbit.

**Documentation**
Include publicly available links to the device documentation. If there are no public links available, please attach the documentation file to the issue.
[Rain Gauge User Manual.pdf](https://github.com/Enapter/marketplace/files/13420089/Rain.Gauge.User.Manual.pdf)
[3S-Rain_Gauge-2.pdf](https://github.com/Enapter/marketplace/files/13420091/3S-Rain_Gauge-2.pdf)

**Questions and Ideas**
If you have any integration questions or implementation ideas that will ensure a high-quality integration, please suggest them here.

anataty, 2023-11-22:
Hi @tjzay! Thank you for providing the device information. I have a couple of questions:
1. Are you planning to use [ENP-RS485](https://handbook.enapter.com/modules/ENP-RS485/ENP-RS485.html) module?
2. Which metrics are you interested in? (I'm referring to pages 9-11 of User Manual)

tjzay, 2023-11-23:
Hi @anataty, 
Thanks for your reply and work.
1. We are not using the ENP-RS485 module as we are using the Arrakis MK4 which does not require it.
2. We are interested in the rain gauge (hour, min, and sec). All the other metrics (air temperature etc.) are not measured by this device .

Thank you!

anataty, 2023-11-24:
Hi @tjzay thank you for the fast answer! We'll start writing this blueprint soon :) I'll keep you updated

Rina-an, 2023-12-04:
Hi @tjzay! 
[Blueprint](https://github.com/Enapter/marketplace/tree/main/analog_io_modules/seven_sensor_rain_gauge) for your device is ready. Drop us a message, if you have any issues

closed 2024-01-09 by anataty