Enapter/marketplace #275: [Temperature sensor, Infrasensing, and BASE-WIRED]

tjzay, 2023-11-21:
**Device Description**
Provide a clear and concise description of the device, and include the link to the device model on the vendor website.
This is a sensor gateway that will be used for one temperature and humidity sensor (ENV-THUM). The sensor connects to the gateway which will then be in communication with Enapter EMS. The gateway also has a temperature sensor on it which we will also be using.
https://infrasensing.com/sensors/sensor_humidity_temperature.asp
https://infrasensing.com/sensors/baseunit.asp

**Device Access**
Do you have access to this device? Will you be able to test the integration?
Yes, it is currently being shipped and will be able to test.

**Communication Protocols**
Specify the communication protocols that the device supports. E.g. Ethernet, RS485, RS232, CAN, DIO, etc.
Ethernet Modbus TCP

**Documentation**
Include publicly available links to the device documentation. If there are no public links available, please attach the documentation file to the issue.
https://infrasensing.com/support/manuals.asp

**Questions and Ideas**
If you have any integration questions or implementation ideas that will ensure a high-quality integration, please suggest them here.

anataty, 2023-11-22:
Hi @tjzay! I've created [PR](https://github.com/Enapter/marketplace/pull/276) with InfraSensing ENV-THUM sensor. Can you please check it out and tell me, if all needed [metrics](https://github.com/Enapter/marketplace/blob/e37348fb76f4ce020d9cc3fd97109acc3340f5bd/analog_io_modules/infrasensing_env-thum/manifest.yml#L38) are present?

tjzay, 2023-11-23:
Hi @anataty. Thanks a lot. I think it all looks great. Only small issue that I can see is that the unit of the humidity is written as celsius but it is actually RH.

anataty, 2023-11-24:
Thank you for noticing that. I fixed it and published the [blueprint](https://marketplace.enapter.com/device/infrasensing_env-thum/infrasensing_env-thum). If you have any further questions/suggestions - I'm happy to help :)

tjzay, 2023-11-28:
Hi @anataty,
Thanks so much. 
I am currently testing the device and there is no signal coming through on the charts. I am using Arrakis Mk4 and have plugged the ethernet cable out of the sensor into one of the LAN ports. I have tested the sensor direct to my computer and communications are working through that. I have also correctly inserted the IP address and MODBUS address. This is my first integration so I am likely doing something wrong but I am not sure what it is. I have attached screenshots. Are you able to help?
Also, we are hoping to add another temperature +humidity sensor to the base sensor (so it will have two of the same temp. and humidity sensors connected). Do we need to change the blueprint for that?
[image]
[image]
[image]
[image]