Enapter/marketplace #287: [Solenoid valve, Airtac, and Solenoid Air Valve - 3V21008NCBG]

tjzay, 2024-01-09:
**Device Description**
This is a normally closed solenoid valve. In this project, it is required to be opened all the time, unless there is a safety issue. https://trimantec.com/products/airtac-3v200-solenoid-valve-3v21008ncbg

**Device Access**
Do you have access to this device? Will you be able to test the integration?
Yes.

**Communication Protocols**
Specify the communication protocols that the device supports. E.g. Ethernet, RS485, RS232, CAN, DIO, etc.
TTL 24 VDC

**Documentation**
Include publicly available links to the device documentation. If there are no public links available, please attach the documentation file to the issue.
https://cdn.shopify.com/s/files/1/0068/3235/7429/files/3V200-Trimantec.pdf

**Questions and Ideas**
If you have any integration questions or implementation ideas that will ensure a high-quality integration, please suggest them here.

nkrasko, 2024-01-09:
Hi @tjzay,

it seems like this valve has no intergration protocols and should be powered with just 24VDC. In this case you can use any of the already supported smart relays: https://marketplace.enapter.com/catalog/relays or propose integration of the smart realy you would like to use.

tjzay, 2024-01-11:
Hi @nkrasko,

Ok, thank you. I'll close this and discuss on discord.

closed 2024-01-11 by tjzay